dawonn/vectornav #7: vectornav Unknown Error

gsp9, 2016-03-11:
Hello,
I am trying to connect my vn200, while doing so, I'm getting an Unknown Error.

````
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://HPi5:43767/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /vectornav/async_output_rate: 100
 * /vectornav/async_output_type: 0
 * /vectornav/binary_data_output_port: 2
 * /vectornav/binary_gps_data_output_rate: 4
 * /vectornav/binary_imu_data_output_rate: 100
 * /vectornav/binary_ins_data_output_rate: 20
 * /vectornav/gps/frame_id: LLA
 * /vectornav/imu/frame_id: LLA
 * /vectornav/serial_baud: 115200
 * /vectornav/serial_port: /dev/ttyUSB0

NODES
  /
    vectornav (vectornav/vn200_node)

auto-starting new master
process[master]: started with pid [9411]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to c267786e-e7cf-11e5-8fd2-142d279346e7
process[rosout-1]: started with pid [9424]
started core service [/rosout]
process[vectornav-2]: started with pid [9431]
[ INFO] [1457731521.694579303]: Initializing vn200. Port:/dev/ttyUSB0 Baud:115200

[ INFO] [1457731521.716068416]: About to set SynchronizationControl
[ INFO] [1457731521.726362555]: Set SynchronizationControl
[FATAL] [1457731536.777613455]: Could not set BinaryResponseListener output on device via: /dev/ttyUSB0, Error Text: Unknown Error
[vectornav-2] process has died [pid 9431, exit code 1, cmd /home/gaurav/catkin_ws/devel/lib/vectornav/vn200_node __name:=vectornav __log:=/home/gaurav/.ros/log/c267786e-e7cf-11e5-8fd2-142d279346e7/vectornav-2.log].
log file: /home/gaurav/.ros/log/c267786e-e7cf-11e5-8fd2-142d279346e7/vectornav-2*.log
````
I tried looking into the source files but couldn't find anything. Can you help me with this?

Thanks
-gsp9

efimiap, 2017-04-10:
Hello,
I have the exact same problem, did anybody find an answer?

Thank you,
Efimia

dawonn, 2017-07-14:
Old.

closed 2017-07-14 by dawonn