dawonn/vectornav #14: Library for vectornav 100

dahlan77, 2018-01-14:
I clone this package and unfortunately .it doesn't support for my vectornav 100.Any solution to make this package can be used with vectrnav 100

Best Regard

Dahlan

dawonn, 2018-01-15:
I'm actually working on vn100 and 300 right now. But it will be a week or two before I can publish a beta version for you

dahlan77, 2018-01-15:
thank you dawonn

for your helping..its really helpfull for me  

Best regard 

Dahlan

dawonn, 2018-02-12:
> I am sorry dawonn
>
> how about to use vectornav 100 for gps
>
> its still really confused mee..because using vectornav 200 is different
>
> Dahlan

VN100 does not support GPS.

dahlan77, 2018-02-12:
really
but in datasheet 

VectorNav | VN-100 SMD | NavtechGP 
VN100 Development Kit

there are part for gps

closed 2018-02-17 by dawonn