dawonn/vectornav #15: VN200 Erratic Readings

acherrera, 2018-05-09:
We have a VN200 mounted on a drone and are using it for mapping. The IMU is supposed to have vibration filtering built in. When I test the system with the Windows GUI, the data looks goods - less than one degree of vibration induced error.

But when I use ROS with this package the readings are very erratic once the motors start. I am curious - does this package read directly from the IMU or is this data taken after the on-board filtering.

We are using the rugged version if that makes a difference. Thanks for the help, hoping this issue is easy to track down.

dawonn, 2018-05-10:
Which ROS Topics are you subscribing to exactly?

dawonn, 2018-05-10:
Looking at the manual, it appears that the IMU binary group is uncompensated sensor data.  You should be able to rather easily change the code to use the AngularRate and Accel data groups to get compensated data streams.  

I'm out of town, but I'll make these changes when I return. If you try this and it works, please send a Pulll Request and I'll merge it in for everyone to enjoy. Thank you for your report!

[image]

dawonn, 2018-05-10:
Scratch that... I *am* using those binary groups....

`COMMONGROUP_TIMESTARTUP | COMMONGROUP_QUATERNION | COMMONGROUP_ANGULARRATE | COMMONGROUP_POSITION | COMMONGROUP_ACCEL | COMMONGROUP_MAGPRES`,
			
I don't see any other message groups that would indicate that they are _more_ compensated... so this may be an issue to bring up with VN...

acherrera, 2018-05-11:
Found the source of the error, has nothing to do with your node - it appears to be working perfectly. Thank you for the help

closed 2018-05-11 by acherrera

Tones29, 2019-12-03:
@acherrera, just being curious, what solved your issue? I have a similar issue.