dawonn/vectornav #16: Heading and Orientation Questions

acherrera, 2018-05-17:
When facing North should the sensor read 0? I appear to be getting a value of 90 deg (pi/2). This is under the vectornav/Mag topic after converting it to an euler reading. 

Second, the Z axis may be flipped on the /vectornav/IMU topic. We are having lots of issues getting this IMU working and one of them is the Z axis points the opposite way of the other IMU's that we have used. I've looked through the source code and it appears that this value is read straight from the sensor, but perhaps you have some feedback on something that I am missing on this. I've applied a transform so that Z axis matches the other sensor but this throws everything else off.

dawonn, 2018-05-18:
Please refer to: http://www.ros.org/reps/rep-0105.html

[image]

If you look at the sensor, there is an X in the center of the Z axis, which indicates that the Z axis does point down, but still follows the right hand rule. If you simply flip the z axis, you now have a left hand coordinate system and ROS will likely be confused. It is likely necessary for you to apply a Transform (TF) to rotate the sensor frame to Z-up X-forward to get the expected results. 

That said,  REP 105 states that 0-deg should point east, and if we rotate the sensor pi/2 that would then point north, which seems correct to me.  

I hope that's helpful; If I have misunderstood, please do not hesitate to reach out. :)

closed 2018-05-18 by dawonn

acherrera, 2018-05-21:
Note that I'm passing the quaternion into a converter to output the orientation in X, Y, Z radians as I find those much easier to read. 

The sensor is mounted "upside down" from what the image shows so the axes align correctly. The test I did below was done on a flat surface just to make sure everything looked right.

I think I narrowed down where these issues are coming from. When the IMU is faced like it is in the image, the orientation returns 0,0,0. However if you flip the IMU so the axes are aligned correctly, the orientation returns 3.14, 0, 0. That is, it reads as if the IMU is rotated 180 degrees on the X axis even though the axes are aligned correctly. I'm unsure what to do about this. 

Once again, please correct me if needed, but I do not believe the reading should be coming back this way.

dawonn, 2018-05-21:
Please server to Section 2.6.1: Sensor Coordinate Frame. It sounds like the device is operating as intended, but you would like the reference frame to be pointing up (Typical for ground based systems).

Section 6.2.4 - Reference Frame Rotation may be of use to you. You can enter a transformation matrix to change the default sensor axis.  

Perhaps the 'best' soution is to subscribe the the MAG topic, rotate the data as you like,  and then republish it for the rest of your application to use. Should be a pretty simple node.