dawonn/vectornav #20: I can not start with launch file, permission denied

OnesoulKang, 2018-08-22:
I just bought VN-200 and I tried to run it in first time. but I got some error from the launch file. I don't know why but something is denied. my error messages are below.

----------------------------------
... logging to /home/hansol/.ros/log/ad0a178a-a5b2-11e8-bbbd-28c2dd8a2fa4/roslaunch-Onesoul-4603.log
Checking log directory for disk usage. This may take awhile.                                        
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Onesoul:38849/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /vectornav/async_output_rate: 40
 * /vectornav/frame_id: vectornav
 * /vectornav/serial_baud: 115200
 * /vectornav/serial_port: /dev/ttyUSB0

NODES
  /
    vectornav (vectornav/vnpub)

ROS_MASTER_URI=http://localhost:11311

process[vectornav-1]: started with pid [4800]
[ INFO] [1534904790.230489915]: Connecting to : /dev/ttyUSB0 @ 115200 Baud                          
terminate called after throwing an instance of 'vn::permission_denied'                              
  what():  Permission denied for item.
[vectornav-1] process has died [pid 4800, exit code -6, cmd /home/hansol/catkin_ws/devel/lib/vectornav/vnpub __name:=vectornav __log:=/home/hansol/.ros/log/ad0a178a-a5b2-11e8-bbbd-28c2dd8a2fa4/vectornav-1.log].
log file: /home/hansol/.ros/log/ad0a178a-a5b2-11e8-bbbd-28c2dd8a2fa4/vectornav-1*.log               
all processes on machine have died, roslaunch will exit                                             
shutting down processing monitor...
... shutting down processing monitor complete
done

dawonn, 2018-08-22:
Take a close look at the error message, you don't have permission to access
the TTY USB serial port. Add yourself to the dialer. And then re login to
your account

OnesoulKang, 2018-08-22:
Thanks! Problem Solved

closed 2018-08-22 by OnesoulKang