dawonn/vectornav #21: How get the data with higher Hz?

OnesoulKang, 2018-08-23:
I tried to get the data with 200hz. But I got an error message that the baudrate is insufficient.

So I changed the rate 115200 to 230400. and I got a timeout error message like below.

[ INFO] [1534993022.308572552]: Connecting to : /dev/ttyUSB0 @ 230400 Baud 
terminate called after throwing an instance of 'vn::timeout'               
  what():  timeout                                                                                  
[vectornav-1] process has died [pid 3406, exit code -6, cmd /home/hansol/catkin_ws/devel/lib/vectornav/vnpub __name:=vectornav __log:=/home/hansol/.ros/log/42469120-a676-11e8-8e8d-28c2dd8a2fa4/vectornav-1.log].
log file: /home/hansol/.ros/log/42469120-a676-11e8-8e8d-28c2dd8a2fa4/vectornav-1*.log

How can i get the data with higher Hz?

dawonn, 2018-08-23:
Perhaps your serial port doesn't support that baud rate.

if you don't need all of the date of that subscribed by default, you can
reduce the amount of data topics that are subscribed over the serial link.
You'll have to modify the code a bit but it's pretty straightforward, just
grab the data items that you actually care about.

OnesoulKang, 2018-08-30:
I asked the company. and their answers below.
----
The VN-200 defaults to 115200 baud, and you will always need to connect at that baud-rate initially unless you change the baud-rate setting on the sensor (through the appropriate Write Register command) and save it to non-volatile memory using the Write Settings command - after which the unit will start-up with the desired baud-rate.
----
It is possible to change the default baud-rate by using sensor explorer which is they provide in windows.

closed 2018-08-30 by OnesoulKang

samkys, 2018-08-31:
@OnesoulKang 
Just an FYI I am working on modifying this driver so that you can set the desired baud rate without writing to non-volatile memory. It is possible, I just need a little more time as its currently in development and not user intuitive. Hopefully will have this done by the end of the day.

samkys, 2018-08-31:
@dawonn 
Have you tried multiple baud rates using the vs.changeBaudRate(baudrate); call on a sensor?

Reason I ask is all of the valid baud rates from the data sheet work except 128000 when trying to update them on my end. I can catch that as an exception but curious if its my IMU or some other thing as even the 128000 is defined in the current library as well as the new one.

samkys, 2018-09-01:
Repo here has something for anyone interested. Would like to get feedback on it as it needs more testing. https://github.com/samkys/vectornav