dawonn/vectornav #24: stuck on line 199 of main.cpp

James-Meyers, 2018-11-13:
This may very well be a simple issue, but I figured I'd ask here. I have roscore running and when I execute "roslaunch vectornav vectornav.launch" in my second terminal, I seem to get stuck on line 199 of main.cpp. I'm able to see the /vectornav/IMU GPS etc. when I use rostopic list, but am unable to conduct rostopic echo of any of them. The outputs of last message from rostopic echo /rosout and rostopic info /vectornav/IMU are below

------------------------------------------------

rostopic echo /rosout

header: 
  seq: 8
  stamp: 
    secs: 1542141873
    nsecs: 984499351
  frame_id: ''
level: 2
name: "/vectornav"
msg: "Model Number: VN-300T-CR"
file: "/home/jmeyers/catkin_ws/src/vectornav/src/main.cpp"
function: "main"
line: 199
topics: [/rosout, /vectornav/IMU, /vectornav/Mag, /vectornav/GPS, /vectornav/Odom, /vectornav/Temp,
  /vectornav/Pres]
---

---------------------------------

rostopic info /vectornav/IMU

Type: sensor_msgs/Imu

Publishers: 
 * /vectornav (http://jmeyers-SAMSUNG:41553/)

Subscribers: None

---------------------------------

rostopic list

/rosout
/rosout_agg
/vectornav/GPS
/vectornav/IMU
/vectornav/Mag
/vectornav/Odom
/vectornav/Pres
/vectornav/Temp

dawonn, 2018-11-13:
That's a great start. It means you are talking to the sensor, but for some
reason it is not configured to output data (is my guess).

You might try to factory reset the sensor and try again. There are many
settings stored in the sensor that could cause problems.

James-Meyers, 2018-11-13:
@dawonn I'll give that a shot and post here if that works, thanks for the quick response!

James-Meyers, 2018-11-13:
So I switched over to my Windows side to do some troubleshooting with Vectornav's default "Sensor Explorer GUI" and whenever I connect the sensor with my COM port as COM5 and baudrate 115200 with the default selection to enable "querying", I don't get live updates on my dynamic graphs for Yaw, Pitch, and Roll. As soon as I uncheck this box (so I disable querying), the Yaw, Pitch, and Roll now plot. When I switched back over to my Ubuntu side and ran ROS, I was able to succesfully use rostopic echo /vectornav/IMU and see the messages! Why this worked, I have no idea, but ended up not having to do a full reset. Thanks again for the help!

closed 2018-11-13 by James-Meyers