dawonn/vectornav #25: Rate mismatch

athtest800, 2019-02-05:
Hello!
I have set "async_output_rate: 200" in the yaml but the rostopic tells that it has average rate 160.
average rate: 160.145
	min: 0.005s max: 0.008s std dev: 0.00044s window: 670

Is it correct to have lower rate published in ROS?

I have edited the main.cpp with the following:

// Configure binary output message
    BinaryOutputRegister bor(
            ASYNCMODE_PORT2,
            1000 / async_output_rate,  // update rate [ms]
            COMMONGROUP_QUATERNION
            | COMMONGROUP_ANGULARRATE
            | COMMONGROUP_POSITION
            | COMMONGROUP_ACCEL
            | COMMONGROUP_MAGPRES,
            TIMEGROUP_NONE,
            IMUGROUP_NONE,
            GPSGROUP_NONE,
            ATTITUDEGROUP_NONE, //<-- returning yaw pitch roll uncertainties
            INSGROUP_NONE);

dawonn, 2019-02-05:
You may be limited by the bandwidth available via the serial port.

You can try reducing the number of messages being sent or increase your
baud rate.

I seem to recall a similar issue at some point...

athtest800, 2019-02-05:
I solved it by changing the line
1000 / async_output_rate
to
800 / async_output_rate

shrijitsingh99, 2019-02-06:
> Do you have any idea why that fixed it?
> […](#)

The second argument to the BinaryOutputRegister constructor represents the rate divisor.
The rate divisor is used to divide the maximum possible rate at which the device can operate.
The output rate = (maximum rate of IMU) / RateDivisor.
So the case of @athtest800 (1000 / 200 ) = 5, so the rate divisor was 5.
The maximum rate at which the VectorNav IMU operate is 800Hz so the rate at which he got the output was actually correct as (800/5) = 160Hz.

This is from the VectorNav manual:

> RateDivisor sets the fixed rate at which the message is sent out the
selected serial port(s). The number given is a divisor of the ImuRate which is nominally 800Hz. For example to have the sensor output at 50Hz you would set the Divisor equal to 16.

The parameter async_output_rate is quite misleading in my opinion and should be changed to rate divisor as in the future if the newer models support higher rates, then the formula would again need to be altered depending on the model.

athtest800, 2019-02-09:
shrijitsingh99 is right...

closed 2019-02-09 by athtest800