dawonn/vectornav #26: GPS data is not publishing

naveenkmedi, 2019-02-08:


dawonn, 2019-02-13:
It should 'just work' as long as you have the correct serial port specified.

shrijitsingh99, 2019-02-17:
Make sure you are getting a fix. It may take the VectorNav upto a minute to lock on to your location.

infracollin, 2019-04-05:
I also am having this issue where /vectornav/GPS and /vectornav/Odom don't print any output with echo. Like naveenkmedi, my IMU, Mag, Pres, and Temp data output fine. I'm using a raspberry pi 3 B+ running Ubuntu 16.04.5 LTS. The serial port used in params/vn100.yaml is /dev/ttyUSB0. I've even had to chmod 777 it for roslaunch to work. I'm using a Linx ANT-GPS-SH2-MMX antenna connected via the MMCX port of the VN-200 rugged.

I have tested connecting to VN gui on Windows 10. (COM 3, baud rate 115200) outside in an open area.
One apparent issue is that even after 5 minutes the Sensor Status window still shows: Mode: 1 (Aligning), 3D Fix Acquired.
Second is, my Sensor Status also doesn't look like the Sensor Status window that is shown in the Help Browser. Mine doesn't show these sets of data: GNSS Satellites, GNSS Compass Startup Status, Current Heading, Highest Satellite CNO, and Current State.

Could this be an issue with the antenna itself?
Is there an NMEA output message I need to configure within the gui first? (Config Register > GNSS > NMEA Output Register)
Thanks for any help.

infracollin, 2019-04-16:
I've been testing from an older commit of this ROS driver. I've experienced good /vectornav/GPS data publishing from the terminal. Now, on a recent pull and rebuild of the ROS driver, rostopic echo shows the same thing naveenkmedi explains.
good data for IMU, Mag, Pres, Temp, but no GPS nor Odom.

From the VN control center gui, I noticed there is method for writing to the VN200 what binary output is required since this is hardcoded. Maybe a previous version of the driver did this operation? I'm wondering if there is a specific configuration that the VN200 needs to be? Does something with the VN200 need preconfigured first? Thanks.

dawonn, 2019-04-24:
Which commit are you using? Can you send me a pull request that restores
the functionality for the current master branch?

-D

“The greater danger for most of us lies not in setting our aim too high and
falling short; but in setting our aim too low, and achieving our mark.” -
Michelangelo

bjajoh, 2019-05-09:
I can confirm the same problem with the VN300. GPS and Odom are missing!

dawonn, 2019-05-11:
Interesting, can you try older commits and see if they are present?

https://github.com/dawonn/vectornav/commits/master/src/main.cpp

dawonn, 2019-06-04:
Merged a 'fix' to not require INS_fix to publish gps data. it will publish whatever it has.

closed 2019-06-04 by dawonn

dlee640, 2021-03-17:
Hello. I am using the most recent commit for my VN100 on Ubuntu 18.04 via ttyUSB0. I get rostopic echo for everything except Odom and GPS topics. I don't think the merge fixed this issue. Can someone help me out on where to fix the code to get both of those to work?

bjajoh, 2021-03-17:
The VN100 is an AHRS without GNSS functionality, so you will have a hard time to get any GPS or Odom topics. ;)

dlee640, 2021-03-17:
That was fast! and Gotcha, so that means VN100 is not meant to publish these topics, since GNSS functionality isn't a part of the option for VN 100? Thank you. I will close my new issue.

bjajoh, 2021-03-17:
I'm speed. :D Yeah, the VN100 ist just Acceleration and Gyro. No position. :)

dawonn, 2021-03-17:
Thanks @bjajoh !!