dawonn/vectornav #28: Odom Messages missing.

bjajoh, 2019-05-11:
Hey,

I have an VN300 and the GPS messages are partially missing, the Odom Messages are not being published at any time.
It must be a problem with this ROS Package.

dawonn, 2019-05-11:
Interesting, can you try older commits and see if they are present? I'm out of the country for the next week, so a pull request would be really great.

https://github.com/dawonn/vectornav/commits/master/src/main.cpp

bjajoh, 2019-05-11:
I analyzed the problem further and found the issue.
I'm not sure why but commenting out "/if (cd.insStatus() == INSSTATUS_GPS_FIX)" and "if (pubOdom.getNumSubscribers() > 0)" solves the problem. It seems like one of them doesn't get into the correct state.

dawonn, 2019-05-11:
My guess is that insstaus is valid, but not in that state. Can you just
remove the insstaus condition, and if that works, send me a pull request?

bjajoh, 2019-05-13:
I commented it out, everything works well and created a pull request.

closed 2019-05-13 by bjajoh