dawonn/vectornav #30: Invalid Operation

smorad, 2019-05-17:
This package appears to be crashing on startup running ROS melodic with the VN200

```
smorad@jnash-Precision-T7600:~/code/catkin_ws/src$ roscore &
...
smorad@jnash-Precision-T7600:~/code/catkin_ws/src$ roslaunch vectornav vectornav.launch
...
process[vectornav-1]: started with pid [9859]
[ INFO] [1558130212.188033718]: Connecting to : /dev/ttyUSB0 @ 115200 Baud
[ INFO] [1558130212.189373668]: Connecting with default at 9600
terminate called after throwing an instance of 'vn::invalid_operation'
  what():  invalid operation
[vectornav-1] process has died [pid 9859, exit code -6, cmd /home/smorad/code/catkin_ws/devel/lib/vectornav/vnpub __name:=vectornav __log:=/home/smorad/.ros/log/a9160b1a-78ee-11e9-a906-f8b1569c5fc8/vectornav-1.log].
log file: /home/smorad/.ros/log/a9160b1a-78ee-11e9-a906-f8b1569c5fc8/vectornav-1*.log
all processes on machine have died, roslaunch will exit
```
Vectornav seems to be functional over usb

```
smorad@jnash-Precision-T7600:~/code/catkin_ws/src$ stat /dev/ttyUSB0 
  File: /dev/ttyUSB0
  Size: 0               Blocks: 0          IO Block: 4096   character special file
Device: 6h/6d   Inode: 682         Links: 1     Device type: bc,0
Access: (0664/crw-rw-r--)  Uid: (    0/    root)   Gid: ( 1000/   jnash)
Access: 2019-05-17 15:01:03.775954898 -0700
Modify: 2019-05-17 15:01:03.775954898 -0700
Change: 2019-05-17 15:01:03.775954898 -0700
 Birth: -
```

dawonn, 2019-05-17:
Maybe you could use GDB to figure out what line the node is crashing on?

This note uses the library from vectren Ave to handle all of the
communication, and I am fairly confident that I tested both the serial
ports on my devices.

smorad, 2019-05-17:
Issue was insufficient permissions. 

`chmod a+rw /dev/ttyUSB0` fixed the problem.

closed 2019-05-17 by smorad