dawonn/vectornav #32: cannot launch vectornav.launch. received sensor error 'InvalidRegister'

sawansaurabh, 2019-06-20:
I'm using vectornav VN-200T Rugged. When i try to launch file vectornav, it exits with the following error. 

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /vectornav/angular_vel_covariance: [0.01, 0.0, 0.0, ...
 * /vectornav/async_output_rate: 40
 * /vectornav/frame_id: imu_link
 * /vectornav/linear_accel_covariance: [0.01, 0.0, 0.0, ...
 * /vectornav/orientation_covariance: [0.01, 0.0, 0.0, ...
 * /vectornav/serial_baud: 115200
 * /vectornav/serial_port: /dev/ttyUSB0
 * /vectornav/tf_ned_to_enu: False

NODES
  /
    vectornav (vectornav/vnpub)
...
[ INFO] [1561021549.214510223]: Connecting with default at 115200
[ INFO] [1561021549.330735692]: Connected baud rate is 115200
[ INFO] [1561021549.356993637]: Device connection established
[ INFO] [1561021549.377947955]: Model Number: VN-200T-CR
**terminate called after throwing an instance of 'vn::sensors::sensor_error'
  what():  received sensor error 'InvalidRegister'**
[vectornav-1] process has died [pid 5668, exit code -6, cmd /home/bci-ubuntu/Desktop/work_station/catkin_ws/devel/lib/vectornav/vnpub __name:=vectornav __log:=/home/bci-ubuntu/.ros/log/66974b06-9332-11e9-9710-0f9cff702591/vectornav-1.log].
log file: /home/bci-ubuntu/.ros/log/66974b06-9332-11e9-9710-0f9cff702591/vectornav-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

I'm unable to figure out what might be causing the error. Kindly, help me with the fix.

P.S: This is the first issue i am raising on github so kindly pardon if i didn't follow the proper norms.

dawonn, 2019-06-20:
Welcome to the community!

I have not seen this error before. Perhaps you can add additional print
statements to figure out exactly where the node is crashing?

sawansaurabh, 2019-06-27:
> Welcome to the community! I have not seen this error before. Perhaps you can add additional print statements to figure out exactly where the node is crashing?
> […](#)

Hello dawonn,
I tried to find out what might be causing this error, I went through cpp and header files but couldn't accomplish much. The sensor seems to be working fine in the VectorNav Control Center Windows app which rules out the possibility of hardware errors.
I was wondering if you could point me to where exactly I should be looking for possible errors and putting the print statements for debugging.
I wanted to know if this package has been tested on VN-200T Rugged. From what I can understand, there might be a possibility that  Register address on this one for some sensors might be different from other models.

ihelal, 2019-07-04:
Hi I am having exactly the same issue with a VN-100S-CR. 

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /vectornav/angular_vel_covariance: [0.01, 0.0, 0.0, ...
 * /vectornav/async_output_rate: 40
 * /vectornav/frame_id: imu_link
 * /vectornav/linear_accel_covariance: [0.01, 0.0, 0.0, ...
 * /vectornav/orientation_covariance: [0.01, 0.0, 0.0, ...
 * /vectornav/serial_baud: 115200
 * /vectornav/serial_port: /dev/ttyUSB0
 * /vectornav/tf_ned_to_enu: False

NODES
  /
    vectornav (vectornav/vnpub)

auto-starting new master
process[master]: started with pid [13115]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7956b61c-9e74-11e9-a2b6-080027086fc3
process[rosout-1]: started with pid [13128]
started core service [/rosout]
process[vectornav-2]: started with pid [13135]
[ INFO] [1562255871.713797534]: Connecting to : /dev/ttyUSB0 @ 115200 Baud
[ INFO] [1562255871.714015799]: Connecting with default at 9600
[ INFO] [1562255872.964965823]: Connecting with default at 19200
[ INFO] [1562255874.238292776]: Connecting with default at 38400
[ INFO] [1562255875.488488751]: Connecting with default at 57600
[ INFO] [1562255876.735440225]: Connecting with default at 115200
[ INFO] [1562255876.846316752]: Connected baud rate is 115200
[ INFO] [1562255876.888812187]: Device connection established
[ INFO] [1562255876.916313647]: Model Number: VN-100S-CR
terminate called after throwing an instance of 'vn::sensors::sensor_error'
  what():  received sensor error 'InvalidRegister'

closed 2020-05-05 by dawonn

wathabs, 2020-08-01:
hello did you solve this problem because i am facing the same problem now?

avcs2019, 2020-12-16:
This is very likely because of hardware issue. I was able to get imu data using asynchronous data mode in the getting_started/ main.cpp example but the vs.writeBinaryOutput1(bor) was throwing the 'vn::sensors::sensor_error'. Luckily I had spare sensors and tested on two of them and it worked without any issue.

SunggooJung, 2021-03-08:
try serial_baud to 921600 in params/vn100.yaml