dawonn/vectornav #34: Invalid async_output_rate

JeanAlexisBoulet, 2019-08-21:
Hi,
I have been doing some test with this ros driver and a VN-100T-CR.
I have been experiencing lower data rate than expected.
For example at the maximum baud rate with the default setting at 200 hz i got this result around 120 hz

started roslaunch server http://jeanalexis-VirtualBox:42141/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /vectornav/angular_vel_covariance: [0.01, 0.0, 0.0, ...
 * /vectornav/async_output_rate: 200
 * /vectornav/frame_id: imu_link
 * /vectornav/linear_accel_covariance: [0.01, 0.0, 0.0, ...
 * /vectornav/orientation_covariance: [0.01, 0.0, 0.0, ...
 * /vectornav/serial_baud: 921600
 * /vectornav/serial_port: /dev/ttyUSB0
 * /vectornav/tf_ned_to_enu: False

NODES
  /
    vectornav (vectornav/vnpub)

auto-starting new master
process[master]: started with pid [11825]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5ec62256-c3b8-11e9-a489-080027b93ea4
process[rosout-1]: started with pid [11838]
started core service [/rosout]
process[vectornav-2]: started with pid [11853]
[ INFO] [1566353225.325856340]: Connecting to : /dev/ttyUSB0 @ 921600 Baud
[ INFO] [1566353225.325916754]: DEBUG Connecting at 200 hz
[ INFO] [1566353225.325973424]: Connecting with default at 9600
[ INFO] [1566353226.530958648]: Connecting with default at 19200
[ INFO] [1566353227.739214086]: Connecting with default at 38400
[ INFO] [1566353228.946924818]: Connecting with default at 57600
[ INFO] [1566353230.154021820]: Connecting with default at 115200
[ INFO] [1566353231.359880146]: Connecting with default at 128000
[ INFO] [1566353231.359960069]: Connecting with default at 230400
[ INFO] [1566353232.570166310]: Connecting with default at 460800
[ INFO] [1566353233.776074619]: Connecting with default at 921600
[ INFO] [1566353233.798978157]: Connected baud rate is 921600
[ INFO] [1566353233.804633858]: Device connection established
[ INFO] [1566353233.809630965]: Model Number: VN-100T-CR

-------------------------------
rostopic hz /vectornav/IMU

subscribed to [/vectornav/IMU]
no new messages
no new messages
no new messages
no new messages
no new messages
average rate: 118.775
	min: 0.006s max: 0.019s std dev: 0.00323s window: 111
average rate: 115.245
	min: 0.000s max: 0.025s std dev: 0.00367s window: 223
average rate: 116.835
	min: 0.000s max: 0.025s std dev: 0.00353s window: 343
average rate: 116.620
	min: 0.000s max: 0.025s std dev: 0.00352s window: 459
average rate: 117.602
	min: 0.000s max: 0.025s std dev: 0.00345s window: 582
average rate: 118.010
	min: 0.000s max: 0.025s std dev: 0.00337s window: 702
^Caverage rate: 118.303
	min: 0.000s max: 0.025s std dev: 0.00334s window: 794

If I edit main.cpp BinaryOutputRegister bor(..
And change de 1000 value for 800 as it seem to be in the user manual everything  seem to work properly.

Can you confirm if I'm right or wrong on that? 
Regards

Jean-Alexis

dawonn, 2019-08-21:
That's not unexpected. The 200 hz is a maximum possible update rate, but by
default the node is subscribed to many device topics which saturates the
serial link. If you reduce the number of data items that are being
transmitted over the serial link, you can get a higher update rate.

JeanAlexisBoulet, 2019-08-21:
Yeah I get it but am I wrong when I said that at main.cpp(207) the 1000 refer to the IMURATE which is by default 800 hz according to the specsheet? I am able to get 200hz by correcting those number(1000 to 800) which in turn set the correct RATE_DIVISOR in the binary output register?
With those setting all the data is correctly receive at 200hz

dawonn, 2019-08-27:
Very cool, I'm glad it's working! If you think this should be the default
value, send me a pull request.

PR #35 (2019-08-29): Feature fix rates
Fixes #34 

As noted in #34, the `IMURATE` included in the node is not the standard (factory default) 800Hz. This causes the actual ROS publish rates to be lower than expected (by 20%). 

This PR adds a parameter for the `IMURATE` which defaults to 800. It also fixes a bug while running with a VN100 by checking the device family before building the INS/Odom messages. Since a VN100 doesn't produce INS data, this particular message production should never be run.
closed 2019-08-29 by dawonn