dawonn/vectornav #41: "InsufficientBaudRate" error after connection established

wyxPatrick, 2020-02-20:
Hi, thank you for sharing this nice package! But I encountered an error and couldn't resolve it. Would you please give me some suggestions?

I am using a VN-200 Rugged. When I try to launch, it gives the following error. Then I tried to set baud rate to 230400, 460800 and 921600. However, none of these worked. In addition, only when baud rate is set to 115200, it will show "Device connection established". But soon after, the error message will be raised. I would appreciate it if you could provide some help!

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /vectornav/angular_vel_covariance: [0.01, 0.0, 0.0, ...
 * /vectornav/async_output_rate: 40
 * /vectornav/fixed_imu_rate: 800
 * /vectornav/frame_based_enu: False
 * /vectornav/frame_id: imu_link
 * /vectornav/linear_accel_covariance: [0.01, 0.0, 0.0, ...
 * /vectornav/orientation_covariance: [0.01, 0.0, 0.0, ...
 * /vectornav/serial_baud: 115200
 * /vectornav/serial_port: /dev/ttyUSB0
 * /vectornav/tf_ned_to_enu: False

NODES
  /
    vectornav (vectornav/vnpub)

ROS_MASTER_URI=http://localhost:11311

process[vectornav-1]: started with pid [21043]
[ INFO] [1582235934.676419442]: Connecting to : /dev/ttyUSB0 @ 115200 Baud
[ INFO] [1582235934.676834407]: Connecting with default at 9600
[ INFO] [1582235935.881357699]: Connecting with default at 19200
[ INFO] [1582235937.094659128]: Connecting with default at 38400
[ INFO] [1582235938.303520098]: Connecting with default at 57600
[ INFO] [1582235939.521716401]: Connecting with default at 115200
[ INFO] [1582235939.577478506]: Connected baud rate is 115200
[ INFO] [1582235939.630014711]: Device connection established
[ INFO] [1582235939.681001172]: Model Number: VN-200S-CR
terminate called after throwing an instance of 'vn::sensors::sensor_error'
  what():  received sensor error 'InsufficientBaudRate'
[vectornav-1] process has died [pid 21043, exit code -6, cmd /home/wyxpatrick/nav_ws/devel/lib/vectornav/vnpub __name:=vectornav __log:=/home/wyxpatrick/.ros/log/4c8ccc1e-5429-11ea-b2ea-4889e71772df/vectornav-1.log].
log file: /home/wyxpatrick/.ros/log/4c8ccc1e-5429-11ea-b2ea-4889e71772df/vectornav-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

dawonn, 2020-02-21:
It's been awhile but I seem to remember not being able to get those to work
either.

I'm pretty sure the vector nav internally supports those higher baud rates,
but most computers do not. This includes the USB to serial adapter that
comes with the dev kit.

Good luck, and if you happen to figure out a solution, relay that back here
for others. :)

wyxPatrick, 2020-02-21:
Thank you Dereck!

Spain2394, 2020-02-25:
I was able to solve this problem by setting my compute ports baud rate to 115200, and the vectornav baud.  1. Configure port: `sudo stty -F /dev/ttyUSB0 115200` 2. Update vn100.yaml  `serial_baud: 23400`

wyxPatrick, 2020-02-25:
@Spain2394 That works perfect on my VN-200! Thank you very much!

A little typo: serial_baud should be set to 230400

closed 2020-02-25 by wyxPatrick