dawonn/vectornav #45: can't roslaunch vectornav vectornav.launch

aleeeexxx, 2020-07-08:
I git clone this repo to my catkin workspace. After I roslaunch, this happened. Any advice or recommendations? Thanks in advance.

process[vectornav-1]: started with pid [12167]
[ INFO] [1594171449.025160510]: Connecting to : /dev/ttyUSB0 @ 115200 Baud
[ INFO] [1594171449.025262354]: Connecting with default at 9600
terminate called after throwing an instance of 'vn::invalid_operation'
  what():  invalid operation
[vectornav-1] process has died [pid 12167, exit code -6, cmd /home/alex/Autonomous-Drifting-Car-Software/catkin_ws/devel/lib/vectornav/vnpub __name:=vectornav __log:=/home/alex/.ros/log/b28fa25a-c0b9-11ea-97d4-0800270a1d5f/vectornav-1.log].
log file: /home/alex/.ros/log/b28fa25a-c0b9-11ea-97d4-0800270a1d5f/vectornav-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

dawonn, 2020-07-08:
I don't have anything particularly insightful for you, other than start
adding print statements in there and figure out exactly where it's
crashing.

Then you can start investigating why. :)

do make sure to let us know what you find, in case the next person has the
same problem.

aleeeexxx, 2020-07-08:
Solved!! It turns out to be a rookie mistake. I was doing it in a virtual machine and didn't add correct USB in virtual machine's setting. You can check your available serial port using `dmesg | grep tty`. If your error is the same with mine (stopping at second [INFO]), is highly likely that you don't connect your USB correctly. Issue #30 could be helpful too.

closed 2020-07-08 by aleeeexxx