dawonn/vectornav #46: No outputs on any rostopics, but also no errors

tianshiz, 2020-09-01:
I get the following when I launch
`[ INFO] [1598998959.066538511]: Connected baud rate is 921600
[ INFO] [1598998959.081410790]: Device connection established
[ INFO] [1598998959.097750150]: Model Number: VN-200S-CR

`
rostopic list shows all the topic but nothing echos out

As a test, I used minicom and can see the ttyUSB0 stream. What's weird is as soon as I launch vectornav, the stream in the minicom stops.

Anyway to debug this?

tianshiz, 2020-09-02:
another thing of note, this older fork from earthmine works for me:
https://github.com/earthmine/vectornav

tianshiz, 2020-09-09:
the issue is ASYNCMODE_PORT1 is configured in main.cpp and I was only using UART2 so nothing was being output on my port. Perhaps would be useful to have this as part of the param?

dawonn, 2020-09-09:
Send me a pull request. :)

closed 2020-10-08 by tianshiz