dawonn/vectornav #47: vn::timeout Error Occuring

advra, 2020-09-19:
This is a common occurrence where I cannot connect to the VN. I am using a VN200 to log data of a period of time.

This is occurred on 3 separate occasions: one when the VN is left connected to the USB port of a raspberry pi for roughly a couple days and twice when our logging program crashed. Below is the error we get:
```
terminate called after throwing an instance of 'vn::timeout'
  what():  timeout
Aborted
```

The immediately solution is to replug the USB so we can connect to it again. Any idea why this is? Ideally I would prefer not to manually reset the USB as the device is remote during long periods of time. I look forward to any suggestions or insight you may provide.

bjajoh, 2022-10-02:
I just had the same and in my case it was usb issue of the devices and has nothing to do with the ros driver.

berndpfrommer, 2022-10-02:
There are ways to reset the USB ports in software. I believe you can even power cycle them.
https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line