dawonn/vectornav #50: Confusion between fixed_imu_rate and async_output_rate

djnugent, 2020-11-17:
I am confused on how to configure the streaming rate of the VN110.

Based on the description in the param file, `fixed_imu_rate` and `async_output_rate` sound like they do the same thing. Looking at the code and the SDK doesn't provide additional clarification. 

I imagine one is supposed to set the output rate and one is to set internal sampling frequency, but treating them in that manner does not yield the expected results. 

It is unclear how to set these 2 values if I want a # hz stream.

dawonn, 2020-11-17:
It's been quite a while since I've used this personally, but if I recall
correctly there's a rate for the raw IMU data which is much higher than the
INS is capable of, so those two data rates are specified independently.

Again, it's been years since I've read the manual so I could be totally on
the wrong track here.  If you want a more concrete answer, take a look at
the source code for the vector nav library itself and figure out exactly
which register is being set for each of those two values. then you can
cross reference to the manual and see what the comments are for each of
those registers.

djnugent, 2020-11-17:
Hi Dereck,
Thanks for the timely response! I was able figure it out. I think I was running into a bandwidth issues(unreported) mixed with confusion about the params. I modified to the code to be more explicit:

renamed `fixed_imu_rate` -> `imu_sample_rate`    # Internal sample frequency of the IMU and max potential output rate
removed `async_output_rate`                                  # Was confusing to have two rates
Added `group1_output_divisor`                                # Transmits messages at imu_sample_rate/group1_output_divisor

closed 2021-03-28 by dawonn