dawonn/vectornav #54: Failed to load nodelet [/imu_filter] of type [imu_filter_madgwick/ImuFilterNodelet] even after refreshing the cache Error

dlee640, 2021-03-18:
Hello.

I am trying to utilize madgwick filter included in the ros package. However, when I launch vn100_imu.launch, it returns 'failed to load nodelet' error. The topics still publish, but I don't think the node for the filtered data is activated as shown below. How can I address this issue?

Thanks.
```
dlee640@dlee640-T580:~/imu_ws$ roslaunch vectornav vn100_imu.launch 
... logging to /home/dlee640/.ros/log/9b234ce6-882a-11eb-8a14-b46bfc2dbb2b/roslaunch-dlee640-T580-13005.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dlee640-T580:36831/

SUMMARY
========

PARAMETERS
 * /imu_data_transformer/target_frame: base_link
 * /imu_filter/gain: 0.01
 * /imu_filter/orientation_stddev: 0.001
 * /imu_filter/publish_tf: False
 * /imu_filter/use_mag: True
 * /imu_filter/world_frame: enu
 * /imu_filter/zeta: 0.001
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /vectornav/angular_vel_covariance: [0.01, 0.0, 0.0, ...
 * /vectornav/async_output_rate: 200
 * /vectornav/fixed_imu_rate: 800
 * /vectornav/frame_based_enu: False
 * /vectornav/frame_id: imu_link
 * /vectornav/linear_accel_covariance: [0.01, 0.0, 0.0, ...
 * /vectornav/orientation_covariance: [0.01, 0.0, 0.0, ...
 * /vectornav/serial_baud: 921600
 * /vectornav/serial_port: /dev/ttyUSB0
 * /vectornav/tf_ned_to_enu: False

NODES
  /
    imu_data_transformer (nodelet/nodelet)
    imu_filter (nodelet/nodelet)
    imu_manager (nodelet/nodelet)
    vectornav (vectornav/vnpub)

auto-starting new master
process[master]: started with pid [13103]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9b234ce6-882a-11eb-8a14-b46bfc2dbb2b
process[rosout-1]: started with pid [13114]
started core service [/rosout]
process[vectornav-2]: started with pid [13121]
process[imu_manager-3]: started with pid [13123]
process[imu_data_transformer-4]: started with pid [13128]
[ INFO] [1616100214.436144725]: Connecting to : /dev/ttyUSB0 @ 921600 Baud
[ INFO] [1616100214.438039890]: Connecting with default at 9600
process[imu_filter-5]: started with pid [13129]
[ERROR] [1616100214.506664752]: Failed to load nodelet [/imu_filter] of type [imu_filter_madgwick/ImuFilterNodelet] even after refreshing the cache: According to the loaded plugin descriptions the class imu_filter_madgwick/ImuFilterNodelet with base class type nodelet::Nodelet does not exist. Declared types are  RobotLocalization/EkfNodelet RobotLocalization/NavSatTransformNodelet RobotLocalization/UkfNodelet SlamGMappingNodelet depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus octomap_color_server/OctomapServerNodelet octomap_server/OctomapServerNodelet pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid pointcloud_to_laserscan/laserscan_to_pointcloud_nodelet pointcloud_to_laserscan/pointcloud_to_laserscan_nodelet stereo_image_proc/disparity stereo_image_proc/point_cloud2 yocs_velocity_smoother/VelocitySmootherNodelet
[ERROR] [1616100214.506716731]: The error before refreshing the cache was: According to the loaded plugin descriptions the class imu_filter_madgwick/ImuFilterNodelet with base class type nodelet::Nodelet does not exist. Declared types are  RobotLocalization/EkfNodelet RobotLocalization/NavSatTransformNodelet RobotLocalization/UkfNodelet SlamGMappingNodelet depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus octomap_color_server/OctomapServerNodelet octomap_server/OctomapServerNodelet pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid pointcloud_to_laserscan/laserscan_to_pointcloud_nodelet pointcloud_to_laserscan/pointcloud_to_laserscan_nodelet stereo_image_proc/disparity stereo_image_proc/point_cloud2 yocs_velocity_smoother/VelocitySmootherNodelet
[FATAL] [1616100214.506928966]: Failed to load nodelet '/imu_filter` of type `imu_filter_madgwick/ImuFilterNodelet` to manager `imu_manager'
[ERROR] [1616100214.529645378]: Failed to load nodelet [/imu_data_transformer] of type [imu_transformer/imu_transformer_nodelet] even after refreshing the cache: According to the loaded plugin descriptions the class imu_transformer/imu_transformer_nodelet with base class type nodelet::Nodelet does not exist. Declared types are  RobotLocalization/EkfNodelet RobotLocalization/NavSatTransformNodelet RobotLocalization/UkfNodelet SlamGMappingNodelet depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus octomap_color_server/OctomapServerNodelet octomap_server/OctomapServerNodelet pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid pointcloud_to_laserscan/laserscan_to_pointcloud_nodelet pointcloud_to_laserscan/pointcloud_to_laserscan_nodelet stereo_image_proc/disparity stereo_image_proc/point_cloud2 yocs_velocity_smoother/VelocitySmootherNodelet
[... cut ...]
[ INFO] [1616100215.641918229]: Connecting with default at 19200
[ INFO] [1616100216.852280678]: Connecting with default at 38400
[ INFO] [1616100218.067426810]: Connecting with default at 57600
[ INFO] [1616100219.284657456]: Connecting with default at 115200
[ INFO] [1616100219.318434143]: Connected baud rate is 921600
[ INFO] [1616100219.342379492]: Device connection established
[ INFO] [1616100219.442383646]: Model Number: VN-100S-CR, Firmware Version: 3.0.0.0
[ INFO] [1616100219.442515652]: Hardware Revision : 7, Serial Number : 100060474

```

lamlaaaam, 2021-06-03:
`sudo apt-get install ros-melodic-imu-transformer`

nkhedekar, 2022-03-09:
This may also be required: sudo apt-get install `ros-melodic-imu-filter-madgwick`. These should be included in the readme