dawonn/vectornav #58: vectornav_msgs ROS2 node does not build without linking tf2_geometry_msgs

jfinken, 2021-04-09:
Sensor details:
- Model: VN-200T-CR
- Firmware Version: 2.0.0.1
- Hardware Version : 3

ROS2 version:
- ros2-foxy built from source

Symptom:
- `colcon build` results in the below header file not found error

Example:
```bash
$ colcon build --symlink-install
...
vectornav/vectornav/src/vn_sensor_msgs.cc:24:10: fatal error: tf2_geometry_msgs/tf2_geometry_msgs.h: No such file or directory
 #include "tf2_geometry_msgs/tf2_geometry_msgs.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
This can be resolved with this change to `CMakeLists.txt` (effectively adding `tf2_geometry_msgs`):
```
diff --git a/vectornav/CMakeLists.txt b/vectornav/CMakeLists.txt
index 0cf42c8..2af957a 100644
--- a/vectornav/CMakeLists.txt
+++ b/vectornav/CMakeLists.txt
@@ -35,7 +35,7 @@ target_link_libraries(${PROJECT_NAME} vncxx )

 # vn_sensor_msgs
 add_executable(vn_sensor_msgs src/vn_sensor_msgs.cc)
-ament_target_dependencies(vn_sensor_msgs rclcpp sensor_msgs vectornav_msgs)
+ament_target_dependencies(vn_sensor_msgs rclcpp sensor_msgs vectornav_msgs tf2_geometry_msgs)
```
I'd be happy to submit a PR to the `ros2` branch if interested.  Thanks!

dawonn, 2021-04-09:
PR's always appreciated!

jjbrl, 2023-06-08:
I will note that a very similar build failure has surfaced due to the include change made in #114 . Older ROS2 distros (ex. Foxy) only provide tf2_geometry_msgs.h and not a .hpp. The solution involves either changing the .hpp include back to a .h on line 24 of vn_sensor_msgs.cc, or manually adding the newer .hpp header to the tf2_geometry_msgs package.

mehradmrt, 2023-08-01:
> 

Thank you, I had the same issue and this helped!

lorenzohess, 2024-10-28:
> I will note that an almost identical build error has surfaced due to the include change made in #114 . Older ROS2 distros (ex. Foxy) only provide tf2_geometry_msgs.h and not a .hpp. The solution involves either changing the .hpp include back to a .h on line 24 of vn_sensor_msgs.cc, or manually adding the newer .hpp header to the tf2_geometry_msgs package.

On commit `2859fe1`, I had to make the following **two** changes:
```
---
 vectornav/src/vn_sensor_msgs.cc  | 2 +-
 vectornav/src/vn_sensor_msgs.hpp | 2 +-
 2 files changed, 2 insertions(+), 2 deletions(-)

diff --git a/vectornav/src/vn_sensor_msgs.cc b/vectornav/src/vn_sensor_msgs.cc
index bcfd1e3..0f42b7f 100644
--- a/vectornav/src/vn_sensor_msgs.cc
+++ b/vectornav/src/vn_sensor_msgs.cc
@@ -16,7 +16,7 @@
 #include <rclcpp_components/register_node_macro.hpp>
 #include <string>
 
-#include "tf2_geometry_msgs/tf2_geometry_msgs.hpp"
+#include "tf2_geometry_msgs/tf2_geometry_msgs.h"
 
 using namespace std::chrono_literals;
 
diff --git a/vectornav/src/vn_sensor_msgs.hpp b/vectornav/src/vn_sensor_msgs.hpp
index 2c32a5d..ab278fa 100644
--- a/vectornav/src/vn_sensor_msgs.hpp
+++ b/vectornav/src/vn_sensor_msgs.hpp
@@ -22,7 +22,7 @@
 #include <sensor_msgs/msg/temperature.hpp>
 #include <sensor_msgs/msg/time_reference.hpp>
 #include <string>
-#include <tf2_geometry_msgs/tf2_geometry_msgs.hpp>
+#include <tf2_geometry_msgs/tf2_geometry_msgs.h>
 #include <vectornav_msgs/msg/attitude_group.hpp>
 #include <vectornav_msgs/msg/common_group.hpp>
 #include <vectornav_msgs/msg/gps_group.hpp>
-- 
```