dawonn/vectornav #63: can't establish connection VN-100

raess1, 2021-05-23:
I am trying this ROS driver and can't establish connection  to the VN-100. 
Specs:
Ubuntu 18.04
/dev/ttyS6
Using UART serial2 from the Vectornav since I have a custom cable for it which I configured the same as serial1 in CC.

Cable:
[image]

Using `picocom -b 115200 /dev/ttyS6` I successfully see data in terminal.

https://user-images.githubusercontent.com/6362413/119269055-10045380-bbf6-11eb-842c-a05329a47707.mp4

Any advice or help would be appreciated!

Thanks in advance
/rf

dawonn, 2021-05-23:
I did not test serial port two on any of my devices, so you might be an
uncharted waters, There's no reason why it shouldn't work however.

You may need to tweak the binary output configuration to point at the other
port:
https://github.com/dawonn/vectornav/blob/634920c046b00bd04ebdbb07ba18bc03174ed5d6/src/main.cpp#L236

raess1, 2021-05-23:
Thanks for the prompt reply. Yeah I did a quick test setting `ASYNCMODE_PORT2` without success. 

So the serial port from vectornav should output binary? Do you have any example from CC how yours is configured over serial1 output? :) That might be helpful

Thanks 
/RF

dawonn, 2021-05-23:
Yes the serial port should be configured as binary output by the Ros node.
This is essentially necessary to get good performance due to the amount of
data being sent. Admittedly the VN100 might be fine because you're only
reading the IMU data but the VN200 and VN300 can definitely saturate the
link.

Essentially that block of code is what does it, There's not too much to it
really. Do you even establish a connection to the device, does it tell you
the serial number and model number etc? If that doesn't work nothing else
is likely to.

You'll also probably want to remove the flags for the output groups that
don't exist on the v100.

raess1, 2021-05-23:
Thanks. Nah it doesn't tell me anything. Not serial or device info. I sell send error image soon.

I guess I can try with the original cable to confirm it's working from that.

I be back

Hämta Outlook för Android<https://aka.ms/AAb9ysg>

raess1, 2021-05-23:
So normal cable gives me  `Device connection established`

But also throws the error 
[image]

raess1, 2021-05-23:
Okej. Increasing baud to` /dev/ttyUSB0 @ 230400 Baud` solves that problem with orginal cable

raess1, 2021-05-24:
So I got my cable working and got Device connection established and Get the following error.
[image]

raess1, 2021-05-24:
success
Changed /dev/ttyS6 @ 921600 Baud
and  `ASYNCMODE_PORT1` -> `ASYNCMODE_PORT2` 

[image]

raess1, 2021-05-24:
So after a little more testing. 
I can't access it after reboot with my cable. It works if unplug my cable and use orginal cable and open CC and then Plugin my cable again and then try the ROS driver again.

raess1, 2021-05-25:
I made a new cable which seems to work alot better. 
[image]

dawonn, 2021-05-25:
That's great, happy to hear it.

closed 2021-05-25 by dawonn