dawonn/vectornav #65: Published topic frequencies not as expected

blaiblayer, 2021-05-27:
Hi. I'm part of a university project where we are using the VN300. I have set the output_sync_rate to 100; however, topics available are not being published at 100Hz (seen with rostopic hz "topic name"). For example vectornav/GPS is published at 80Hz, whilst vectornav/Odom is published at 24Hz. I have checked in the Vectornav Control Centre and the output is at the chosen rate (100Hz), which means the problem must be within the ROS package in Ubuntu 16.04.  Does anyone have the same problem, or know the reason why it is happening? Thans in advance.

alexglzg, 2021-06-08:
I have the same problem, but I have not found any solution yet

TheGermanKid, 2021-06-11:
Similiar issue with the VN-310. It's publishing both INS and IMU messages at 40 Hz while GPS is published at 80 Hz.
Could someone confirm that line 237 in main.cpp uses the correct value? From the VN300 manual: 

> Sets the fixed rate at which the message is sent out the
selected serial port(s). The number given is a divisor of the
ImuRate which is nominally 400 Hz. For example to have the
sensor output at 50Hz you would set the Divisor equal to 8.

But the value used is IMU rate / async rate (default 800/40, so it should result in 20 Hz), For the NMEA Output Register this would be a reasonable value. 

Maybe I am missing/misunderstanding something.

jonasruchti, 2021-06-11:
I had the same problem with a VN300.

I've changed fixed_imu_rate parameter in the yaml file to 400 and disabled Async Output with Vectornav Control Center.

[image]

Now the output rates are as expected.

PR #70 (2021-07-14): Updates and publish rate
closes #65

This merge request will change the following things:

- [x] Update the vectornav programing library to the newest version (1.2.0.0).
- [x] Add a Dockerfile
- [x] Fix issue #65

I was having some issues with my device with respect to publishing rates and it turned out it was related to #65. The issue was that it seems there are two different channels to get the device to send data asynchronously to the host:

1) Async output data: This configures the device to send out data at a fixed rate. This is discussed in chapter 6.2.7 and 6.2.8 of the manual. It's rat [... cut]
closed 2021-07-14 by dawonn