dawonn/vectornav #67: terminate called after throwing an instance of 'vn::invalid_operation'

anayghatpande, 2021-06-04:
I am having vectornav package installed on robotnik robot (melodic).
while launching the vectornav I am getting this error.

setting /run_id to f7aab014-c554-11eb-88bb-0016ebe34540
process[rosout-1]: started with pid [18181]
started core service [/rosout]
process[vectornav-2]: started with pid [18188]
[ INFO] [1622825430.155672335]: Connecting to : /dev/ttyUSB_IMU @ 921600 Baud
[ INFO] [1622825430.156325921]: Connecting with default at 9600
terminate called after throwing an instance of 'vn::invalid_operation'
  what():  invalid operation
[vectornav-2] process has died [pid 18188, exit code -6, cmd /home/rb1/catkin_ws/devel/lib/vectornav/vnpub __name:=vectornav __log:=/home/rb1/.ros/log/f7aab014-c554-11eb-88bb-0016ebe34540/vectornav-2.log].
log file: /home/rb1/.ros/log/f7aab014-c554-11eb-88bb-0016ebe34540/vectornav-2*.log

I even tried reinstalling the vectornav from git clone. still it is not fixed. I checked the solutions from #45 and #30  for permissions and checking serial port. both things are working still vectornav is not launching.
with dmesg | grep tty

[    3.157990] usb 2-2: FTDI USB Serial Device converter now attached to ttyUSB0

serial port info =
rb1@rb1-desktop:~$ udevadm info -a /dev/ttyUSB0 | grep serial
    SUBSYSTEMS=="usb-serial"
    ATTRS{serial}=="FT97PH8D"
    ATTRS{serial}=="0000:00:14.0" 

Let me know what else is possible to solve this.

dawonn, 2021-06-04:
It looks like it the sensor was not found on that serial port at any
supported baud rate. Make sure that you can open up the serial ports and
see data coming in from it manually using minicom for example.

dawonn, 2021-06-04:
Also this is a bit suspicious that it's trying to connect to
dev/ttyUSB_IMU but
then you showed info about dev/ttyUSB0. Check the launch file or launch
parameters and make sure that you're pointing to the right device

anayghatpande, 2021-06-11:
> Also this is a bit suspicious that it's trying to connect to dev/ttyUSB_IMU but then you showed info about dev/ttyUSB0. Check the launch file or launch parameters and make sure that you're pointing to the right device
> […](#)

Actually ttyUSB_IMU is a SYMLINK defined in udev rules. it is given by their documentation. so it actually refers to ttyUSB0. 
udev rules file is ​/etc/udev/rules.d/50-pixhawk.rules​ :
KERNEL​==​"ttyUSB[0-9]*"​, OWNER=​"rb1"​, GROUP=​"dialout"​, MODE=​"0666"
KERNEL​==​"ttyUSB[0-9]*"​, ATTRS{idProduct}==​"6001"​,
ATTRS{serial}==​"​FT97PH8D​"​, NAME=​"%k"​, SYMLINK=​"ttyUSB_IMU"​,
GROUP=​"dialout"​, MODE=​"0666"

same as showing with grep command :  
udevadm info -a /dev/ttyUSB0 | grep serial
    SUBSYSTEMS=="usb-serial"
    ATTRS{serial}=="FT97PH8D"
    ATTRS{serial}=="0000:00:14.0"

So I don't find any issue here. Please let me know if I am missing something.