dawonn/vectornav #73: Odom != ECEF?? Why is an ECEF Position parsed into the Odom frame?

bjajoh, 2021-07-17:
Hi,

for some reason the ECEF Position is used for the odom message. However this doesn't make any sense in my opinion, because ECEF is completely different from the UTM Coordinate System which should be used.
This is not conform with the ROS Odom message or in general anyone who would use it.

Is there any reason for it?

dawonn, 2021-07-17:
Please feel free to send a PR; using ECEF directly doesn't really make sense for the Odom frame....

bjajoh, 2021-07-17:
Ok thank you, yes I'll think about a fix.

dawonn, 2021-07-17:
ECEF is probably a fine choice for a data source, but generate a transform to an ENU frame with the origin at the starting location and translate the pos into that frame. 

Thank you for the help! I don't regularly use this interface obviously. ;)

CollinNHays, 2021-11-17:
Take a look at http://wiki.ros.org/enu. Since the VectorNav cannot provide an ENU/NED position this would be a good example of how to calculate an ENU position with the data the VectorNav provides.
It may be worth implementing an LLA to ENU conversion in the odom topic of this driver.
I may even make a PR out of this if wanted.

dawonn, 2021-11-17:
PR's are very welcome. :)