dawonn/vectornav #75: VN310 support? Dual GNSS

pringithub, 2021-07-27:
Hello!

We have a VN310 and are wondering if this library could add Dual GNSS support as specified in the VN310 spec. 
Thankfully, we are able to use the vn210.yaml file to get GPS data with a single antenna, but would like the option of dual antennae as well.

pringithub, 2021-07-27:
I also don't see any documentation for a 'compass' member variable in any of the rosmsgs, but I do see a reference to getting compass data in #57

dawonn, 2021-07-27:
You should be able to easily support the dual GPS receiver with minimal
modifications to main.cc. hardware support should be there already.

I'll note that this note does not publish the raw GPS data by default,
instead it relies on the INS solution which on the VN 310 should use both
GPS antennas.

It's probably not very helpful but the ROS2 version supports it out of the
box, just need to configure the node parameters to enable the messages you
care about.

MattArceri, 2021-07-29:
The yaw value reported by the VN-310 is aligned with North already. That direction is determined automatically using the dual GPS (as long as you setup the antenna positions properly already). You can just use the yaw value for heading. The compass status registers aren't needed for normal operation of the VN-310.

umiklos, 2023-11-14:
@pringithub did you find a solution? We also want to use the dual GNSS with ROS1.