dawonn/vectornav #77: Set antenna offsets

DavidTMuc, 2021-08-16:
Hello,

I have a VN-300 Dual GNSS and wondering if I could set the Antenna positions directly in your code, instead of the VectorNavs control center??

Thanks!!

DavidTMuc, 2021-08-16:
[image]
[image]

Here are the settings in the Control Center as an example.

dawonn, 2021-08-16:
The ROS2 driver supports this via parameters, but the ROS1 version
doesn't.  You'll have to open up main.cbp and add a little section to set
those parameters, but it should be pretty straightforward, don't be too
afraid. ;)

DavidTMuc, 2021-08-18:
It finally worked!! In the following the code:
// Set Antenna position
    vec3f antenna_position(1.1, 2.2, 3.3);
    vs.writeGpsAntennaOffset(user_data.antenna_position, false);
    vec3f base_line_position(1.1, 2.2, 3.3);
    vec3f base_line_uncertainity(0.1, 0.2, 0.3);
    vs.writeGpsCompassBaseline(user_data.base_line_position,user_data.base_line_uncertainity, false);
    vs.writeSettings();
    
Now I wonder how I could refer to this in the launch file. Could you give me an advice - especially regarding the node handle, which I have to use to solve this issue, I guess?
 
 Thank you!!

dawonn, 2021-08-18:
Yep, use the ROS parameter system to read in those parameters from a launch
file. You can see some examples where I have a parameter for the serial
port and baud rate.