dawonn/vectornav #89: source tree lacks code formatting style file

berndpfrommer, 2021-12-28:
Could one of the maintainers please post a formatting style file? I believe this is the standard ROS one:

```
---
Language: Cpp
BasedOnStyle: Google

AccessModifierOffset: -2
AlignAfterOpenBracket: AlwaysBreak
BraceWrapping:
  AfterClass: true
  AfterFunction: true
  AfterNamespace: true
  AfterStruct: true
BreakBeforeBraces: Custom
ColumnLimit: 100
ConstructorInitializerIndentWidth: 0
ContinuationIndentWidth: 2
DerivePointerAlignment: false
PointerAlignment: Middle
ReflowComments: false
...

```

dawonn, 2021-12-28:
Please send a PR 🙏

dawonn, 2021-12-28:
Make sure you exclude the vncpp library, we don't want to make changes there unless absolutely required.

closed 2021-12-30 by berndpfrommer