dawonn/vectornav #117: Question: Running at higher than 200 Hz

motiv-ncb, 2023-03-07:
Hi, sorry this is a question and not an issue with the package, but I didn't know how else to contact you. 
First, thank you so much for creating this, it has worked fantastic for our SLAM applications.

My question is about the rep rate being limited to 200 Hz in the yaml config file. 
From what I understand from the datasheet, the VN-100 and VN-200 can output EKF data at 400 Hz. 
Currently, we run it at 200 Hz, since that is the highest async_output_rate I can specify in the yaml config. 
Specifically what I want is the quaternion data at 400 Hz, if possible. 

Is it possible to get the quaternion data at 400 Hz using this ROS package?

Thanks very much!

motiv-ncb, 2023-03-07:
As a follow up, am I correct in my understanding that among these listed below, ATTITUDEGROUP_YPRU is the bottleneck, as it comes from the NavFilter module, which runs at 200 Hz? If I remove it from this list, can I then set the output rate to 400 Hz? 

What I need are COMMONGROUP_QUATERNION, COMMONGROUP_ANGULARRATE, COMMONGROUP_ACCEL, and COMMONGROUP_TIMESTARTUP

  BinaryOutputRegister bor(
    ASYNCMODE_PORT1,
    SensorImuRate / package_rate,  // update rate [ms]
    COMMONGROUP_QUATERNION | COMMONGROUP_YAWPITCHROLL | COMMONGROUP_ANGULARRATE |
      COMMONGROUP_POSITION | COMMONGROUP_ACCEL | COMMONGROUP_MAGPRES |
      (user_data.adjust_ros_timestamp ? COMMONGROUP_TIMESTARTUP : 0),
    TIMEGROUP_NONE | TIMEGROUP_GPSTOW | TIMEGROUP_GPSWEEK | TIMEGROUP_TIMEUTC, IMUGROUP_NONE,
    GPSGROUP_NONE,
    ATTITUDEGROUP_YPRU,  //<-- returning yaw pitch roll uncertainties
    INSGROUP_INSSTATUS | INSGROUP_POSECEF | INSGROUP_VELBODY | INSGROUP_ACCELECEF |
      INSGROUP_VELNED | INSGROUP_POSU | INSGROUP_VELU,
    GPSGROUP_NONE);

dawonn, 2023-03-07:
I'm glad that this driver has been helpful for you.

The project is structured to make it easy to get started, and then provide you a foundation to extend and modify for your particular needs.

Been a while since I've personally used this, but IIRC the uart bandwidth becomes a bottleneck for the higher data rates. It's very likely that you will need to disable some of the other message types in order to have enough bandwidth.

You may even need to use a second BOR for the IMU at a higher rate for the raw IMU, and then use the first BOR for the lower rate data.

berndpfrommer, 2023-03-07:
I think it's not providing exactly the data you are asking for, but there is a sample config file in the ROS2 branch for the VN100 running at 800Hz. https://github.com/dawonn/vectornav/blob/aa890673b5e56165d20a9c137120542ca6a36e1e/vectornav/config/vn_100_800hz.yaml