dawonn/vectornav #121: Bug | Orientation is in the wrong coordinate frame

nakai-omer, 2023-05-22:
Hello,

We are using a VN-100T, and we are getting a `-π` roll when the IMU is flat on a surface. In other words, Z is pointing down.
[image]
From looking a little at the code we found a couple of issues:
1) https://github.com/dawonn/vectornav/blob/933af893ab91986110dab2528a6fa6c913f4f932/vectornav/src/vn_sensor_msgs.cc#L181 - There is always a conversion to ENU even if `use_enu` is false. 
2) The above conversion actually breaks the orientation and causes the above mentioned issue. From our tests, the quaternion that arrives is already in ENU, so this conversion is redundant (nevertheless a conversion should exist is `use_enu` is false to turn it into NED).
3) Pose is always converted to ECEF, and afaik should also be in ENU
https://github.com/dawonn/vectornav/blob/933af893ab91986110dab2528a6fa6c913f4f932/vectornav/src/vn_sensor_msgs.cc#L309

I don't mind issuing a PR, just would like to get assurance I am not missing something.

dawonn, 2023-05-22:
Good eye, PRs are appreciated.

nakai-omer, 2023-05-23:
@dawonn create PR #122

AngeLoGa, 2023-11-30:
Greetings, I have also been experiencing this issue when we bought a robot using a VectorNav IMU. My question is, seems that the PR was merged into ros2 branch. Have these changes been made on master branch as well? I'm currently working in Melodic but planning to update to Noetic...thanks from advance.

PR #122 (2023-07-03): Fix orientation and pose for enu
fixed #121