dawonn/vectornav #139: cant get INS messages

abdlrhmn58, 2024-05-26:
iam using ubuntu 18.04 on jetson nano with ROS2 Eloquent
iam using vectornav100
when i ros2 topic echo /vectornav/raw/ins 
it gives me zeroes how to modify it to get real data?
thankyou

dawonn, 2024-05-26:
Could be so many different things, but I would start by making sure you
have the USB port configured correctly and with proper permissions.

Try adding some breakpoints to the node or print statements and figure out
where it's broken. It's a great skill to learn and this driver is pretty
simple to learn on. ;)

Good luck!

abdlrhmn58, 2024-05-28:
@dawonn thank you for your reply 

i make sure before executing the node to write the following command sudo chmod 666 /dev/ttyUSB0

and it works properly when i echo topics like vectornav/raw/common or /vectornav/imu

but what i need is ins when i try to echo /vectornav/raw/ins

[image]
[image]

i tried to search in files vectornav.cc and vn_sensor_msgs.cc

it seems that there is a missing line in publishers at vn_sensor_msgs.cc for ins 

[image]
[image]

[image]
[image]

so what is your opinion ..

i appreciate your help

dawonn, 2024-05-29:
Vm100 doesn't have an INS solution, it has no GNDS receiver.

abdlrhmn58, 2024-05-29:
So its my mistake 
Thanks :)

closed 2024-05-29 by abdlrhmn58